sonyhome/FAB_LED #7: ESP8266 support

dpfavand, 2016-05-01:
Any thoughts on on whether FAB_LED could be made to work on an ESP8266? Particularly looking at the NODEMCU devkits, which expose most of the pins. The Adafruit library is supposed to work (per http://www.instructables.com/id/ESP8266-controlling-Neopixel-LEDs-using-Arduino-ID/ ).

Installing the FAB_LED library and compiling the example AA_SimpleDemo against the NODEMCU device results in a slew of compile errors, most seeming to relate to digitalWriteFast.

Unfortunately I'm not familiar with this level of programming, so I can't investigate more deeply at the moment. Having a lighter-weight library on the versatile but resource-limited ESP8266 chip would be incredibly useful, though. Let me know if there's any way I can help.

sonyhome, 2016-05-03:
Hi.

Failure status:

The ESP8266 uses a Tensilica CPU: If so it's not surprising everything barfs.

I'd love to support it, especially the all-in-one with wifi would make it super cool to drive LEDs.
However I need two major things:
* Knowledge to access the ports to bitbang OR write via buffer to it
* Knowledge to create properly timed wait loops for bitbanging OR a way to configure some hardware to spit out the right high/low pattern durations

After that having proper defines to detect the chipset and all would be great but can always be done manually.

Schedule:

I have a personal need for 2D support, so I'm going to focus on that first.

I also want to do the validation of the support for sk6812 which I'm told works but
I have not tested myself (my chips don't work and it could be soldering since I
can't find for certain the notch on the chip).

I'd like to get ARM/Arduino working next (I need to grab the hardware)

Now, past that:
It looks like this board can be used in the Arduino IDE environment and works with NeoPixels:
http://www.instructables.com/id/ESP8266-controlling-Neopixel-LEDs-using-Arduino-ID/
Hence from there I can probably make it work if it's true by looking into that.

Ha e you tried that and does it work with the NeoPixel library?

Action item:

I'll leave this RFE open.
I'll wait for confirmation the board compiles on the Arduino IDE and there's one or more libraries that work.
I'll also work on it only if I can have someone test the code works for them when I do the dev.

Thanks!

dpfavand, 2016-05-04:
Thanks for the response!

To answer one of your questions - the board does compile on the Arduino IDE (using https://github.com/esp8266/Arduino/), I haven't tried it with the NeoPixels yet, but hope to soon.

However, since posting I have discovered https://github.com/Makuna/NeoPixelBus, which seems to have most of the features I am looking for - I will try it when I get my next strip of pixels. Crucially, it provides 2D support (in a variety of configurations). I'm not sure how the architecture of that library compares to yours, but it might provide some insights at least. So, depending on how well that library performs, it may not be particularly useful to add esp8266 support to your library.

That said, if you decide to go ahead, I could potentially test the code in the future, it will depend on where my projects are at that time.

Cheers

sonyhome, 2016-05-04:
OMG... I overwrote your message by accident :/ I didn't know I could edit someone's response. :(

Can you reply by putting back the couple of links and infos you had? I'm not sure I can undo my mistake.

dpfavand, 2016-05-04:
Haha wow I didn't know that was possible either! Anyway, fortunately I still had the window open - so here's the original in full:

---

Thanks for the response!

To answer one of your questions - the board does compile on the Arduino IDE (using https://github.com/esp8266/Arduino/), I haven't tried it with the NeoPixels yet, but hope to soon.

However, since posting I have discovered https://github.com/Makuna/NeoPixelBus, which seems to have most of the features I am looking for - I will try it when I get my next strip of pixels. Crucially, it provides 2D support (in a variety of configurations). I'm not sure how the architecture of that library compares to yours, but it might provide some insights at least. So, depending on how well that library performs, it may not be particularly useful to add esp8266 support to your library.

That said, if you decide to go ahead, I could potentially test the code in the future, it will depend on where my projects are at that time.

Cheers

sonyhome, 2016-05-04:
Thanks for your response:

I would appreciate if you test out those libraries that you come back here and give me your opinions and what you decided to use. I will then look at those and make a call...

Cheers.

dpfavand, 2016-05-04:
Will do!

sonyhome, 2016-05-04:
I just noticed I also get the thread by email so I had it saved :) Anyways I cleaned up the cruft messages.

This library for me is an exercise in optimized C++ code generation while not short changing functionality and ease of use. On controllers with lots of RAM it's a bit more moot, but if I can I'd like the library to be ubiquitous too :)