sonyhome/FAB_LED #15: Example Code produces lots of compiling errors

guenniman, 2016-09-10:
I tried to compile and upload the raw24Bit example but it just threw errors at me.

Here are the error messages:

In file included from test.ino:1:0:
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:348:2: error: #error "Unsupported Architecture"
 #error "Unsupported Architecture"
  ^
In file included from test.ino:1:0:
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:69:23: error: static data member ‘<anonymous struct>::type’ in unnamed class [-fpermissive]
  static const uint8_t type = PT_RGB;
                       ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:68:16: note: unnamed class defined here
 typedef struct {
                ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:77:23: error: static data member ‘<anonymous struct>::type’ in unnamed class [-fpermissive]
  static const uint8_t type = PT_GRB;
                       ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:76:16: note: unnamed class defined here
 typedef struct {
                ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:84:23: error: static data member ‘<anonymous struct>::type’ in unnamed class [-fpermissive]
  static const uint8_t type = PT_BGR;
                       ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:83:16: note: unnamed class defined here
 typedef struct {
                ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:92:23: error: static data member ‘<anonymous struct>::type’ in unnamed class [-fpermissive]
  static const uint8_t type = PT_RGB | PT_XXXW;
                       ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:91:16: note: unnamed class defined here
 typedef struct {
                ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:101:23: error: static data member ‘<anonymous struct>::type’ in unnamed class [-fpermissive]
  static const uint8_t type = PT_GRB | PT_XXXW;
                       ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:100:16: note: unnamed class defined here
 typedef struct {
                ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:110:23: error: static data member ‘<anonymous struct>::type’ in unnamed class [-fpermissive]
  static const uint8_t type = PT_BGR | PT_BXXX;
                       ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:109:16: note: unnamed class defined here
 typedef struct {
                ^
In file included from test.ino:1:0:
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h: In constructor ‘avrBitbangLedStrip<high1, low1, high0, low0, minMsRefresh, dataPortId, dataPortPin, clockPortId, clockPortPin, colors, protocol>::avrBitbangLedStrip()’:
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:654:41: error: there are no arguments to ‘SET_DDR_HIGH’ that depend on a template parameter, so a declaration of ‘SET_DDR_HIGH’ must be available [-fpermissive]
    SET_DDR_HIGH(dataPortId,  dataPortPin);
                                         ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:654:41: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:655:42: error: there are no arguments to ‘SET_DDR_HIGH’ that depend on a template parameter, so a declaration of ‘SET_DDR_HIGH’ must be available [-fpermissive]
    SET_DDR_HIGH(clockPortId, clockPortPin);
                                          ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:656:41: error: there are no arguments to ‘SET_PORT_LOW’ that depend on a template parameter, so a declaration of ‘SET_PORT_LOW’ must be available [-fpermissive]
    SET_PORT_LOW(dataPortId,  dataPortPin);
                                         ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:657:42: error: there are no arguments to ‘SET_PORT_LOW’ that depend on a template parameter, so a declaration of ‘SET_PORT_LOW’ must be available [-fpermissive]
    SET_PORT_LOW(clockPortId, clockPortPin);
                                          ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:660:28: error: there are no arguments to ‘FAB_DDR’ that depend on a template parameter, so a declaration of ‘FAB_DDR’ must be available [-fpermissive]
    FAB_DDR(dataPortId, 0xFF); // all pins out
                            ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:661:29: error: there are no arguments to ‘FAB_PORT’ that depend on a template parameter, so a declaration of ‘FAB_PORT’ must be available [-fpermissive]
    FAB_PORT(dataPortId, 0x00); // all pins low
                             ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:665:40: error: there are no arguments to ‘SET_DDR_HIGH’ that depend on a template parameter, so a declaration of ‘SET_DDR_HIGH’ must be available [-fpermissive]
    SET_DDR_HIGH(dataPortId, dataPortPin);
                                        ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:666:42: error: there are no arguments to ‘SET_DDR_HIGH’ that depend on a template parameter, so a declaration of ‘SET_DDR_HIGH’ must be available [-fpermissive]
[... 495 lines cut ...]
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:1128:38: error: ‘DELAY_CYCLES’ was not declared in this scope, and no declarations were found by argument-dependent lookup at the point of instantiation [-fpermissive]
    DELAY_CYCLES(low0 - cbiCycles - 20);
                                      ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:1121:34: note: ‘DELAY_CYCLES’ declared here, later in the translation unit
    DELAY_CYCLES(high0 - sbiCycles);
                                  ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h: In instantiation of ‘static void avrBitbangLedStrip<high1, low1, high0, low0, minMsRefresh, dataPortId, dataPortPin, clockPortId, clockPortPin, colors, protocol>::spiSoftwareSendBytes(uint16_t, const uint8_t*) [with int high1 = 8; int low1 = 2; int high0 = 2; int low0 = 4; long unsigned int minMsRefresh = 20ul; avrLedStripPort dataPortId = (avrLedStripPort)4u; unsigned char dataPortPin = 13u; avrLedStripPort clockPortId = (avrLedStripPort)1u; unsigned char clockPortPin = 0u; pixelFormat colors = (pixelFormat)2u; ledProtocol protocol = (ledProtocol)1u; uint16_t = unsigned int; uint8_t = unsigned char]’:
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:820:37:   required from ‘static void avrBitbangLedStrip<high1, low1, high0, low0, minMsRefresh, dataPortId, dataPortPin, clockPortId, clockPortPin, colors, protocol>::sendBytes(uint16_t, const uint8_t*) [with int high1 = 8; int low1 = 2; int high0 = 2; int low0 = 4; long unsigned int minMsRefresh = 20ul; avrLedStripPort dataPortId = (avrLedStripPort)4u; unsigned char dataPortPin = 13u; avrLedStripPort clockPortId = (avrLedStripPort)1u; unsigned char clockPortPin = 0u; pixelFormat colors = (pixelFormat)2u; ledProtocol protocol = (ledProtocol)1u; uint16_t = unsigned int; uint8_t = unsigned char]’
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:1212:44:   required from ‘static void avrBitbangLedStrip<high1, low1, high0, low0, minMsRefresh, dataPortId, dataPortPin, clockPortId, clockPortPin, colors, protocol>::sendPixels(uint16_t, const uint8_t*) [with int high1 = 8; int low1 = 2; int high0 = 2; int low0 = 4; long unsigned int minMsRefresh = 20ul; avrLedStripPort dataPortId = (avrLedStripPort)4u; unsigned char dataPortPin = 13u; avrLedStripPort clockPortId = (avrLedStripPort)1u; unsigned char clockPortPin = 0u; pixelFormat colors = (pixelFormat)2u; ledProtocol protocol = (ledProtocol)1u; uint16_t = unsigned int; uint8_t = unsigned char]’
test.ino:21:39:   required from here
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:856:42: error: ‘SET_PORT_LOW’ was not declared in this scope
    SET_PORT_LOW(clockPortId, clockPortPin);
                                          ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:858:42: error: ‘SET_PORT_HIGH’ was not declared in this scope
     SET_PORT_HIGH(dataPortId, dataPortPin);
                                          ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:860:41: error: ‘SET_PORT_LOW’ was not declared in this scope, and no declarations were found by argument-dependent lookup at the point of instantiation [-fpermissive]
     SET_PORT_LOW(dataPortId, dataPortPin);
                                         ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:856:42: note: ‘SET_PORT_LOW’ declared here, later in the translation unit
    SET_PORT_LOW(clockPortId, clockPortPin);
                                          ^
/home/erik/sketchbook/libraries/FAB_LED/FAB_LED.h:862:43: error: ‘SET_PORT_HIGH’ was not declared in this scope
    SET_PORT_HIGH(clockPortId, clockPortPin);
                                           ^

sonyhome, 2017-01-22:
It looks like your CPU is not supported.

I have not yet fully implemented ARM support, only AVR.
I think APA102 + ARM should work but I need to do some testing.

Sorry :)

closed 2017-01-22 by sonyhome

sonyhome, 2017-07-17:
Note: enhanced how the library fails with invalid CPUs with clearer messages.