sonyhome/FAB_LED #17: Can't seem to get the library to work on an Arduino Mega 2560

healthyuncle, 2016-11-17:
I can get the same installation to work great on an uno, but i ran out of memory after 250 LED's (sk6812). So i upgraded to the Mega, and i cannot get the library to work. Any suggestions?

Thanks so much for the fantastic library!

jonaglon, 2016-12-16:
Hi, I'm also using a Mega2560 and having problems. I am using both ws2812b and sk6812 and both strips work fine with AdaFruit libraries. I really want to use this library though, I trying to control rgbw lights in time to a beat set using midi in captured through the Serial port (usb). I think because of interrupt issues the other libraries mess up the timing from the serial port. I'd love to get this library working - does anyone have it working on Mega?

sonyhome, 2017-01-22:
That's odd. The mega I believe is also an AVR CPU so it should work no problem.

If anything, maybe an issue with the pinout, and the signal going to a pin that is not the one you think.
If you are still interested please contact me to get to the bottom of this.

By the way, you can use palettes to save memory and control more pixels.

DooMMasteR, 2017-03-26:
I am also affected, usind SK6812 on an MEGA wont work.

sonyhome, 2017-03-27:
I will need more infos on what's not working, like actual source code that you think should work but isn't... Without the board nor code I have nothing to work with.

DooMMasteR, 2017-03-27:
I have not had the time so far too look at the code of the lib, but I might think it could be related to the Timer usage on the MEGA, which is different to the normale Arduino... otherwise I can only say that I tried quite some Pins, 2,6,12,22,45 and so on and non worked.
The code however is compiling fine, so it must be something lib related special to the Arduino MEGA :)

sonyhome, 2017-03-30:
How did you specify the port in your source code? maybe that's the issue?

BTW, this library disables the interrupts and timer during transmission, so unless something else is going on... like maybe the chip is not 16mhz but slightly different from the IDE built in board frequency definitions?

sonyhome, 2017-03-30:
If I can find someone with that board willing to experiment a bit we might get lucky.

mattarth, 2017-05-06:
I'm trying to use FAB_LED with a mega, and I'd be happy to experiment to make it work if I can help

sonyhome, 2017-05-08:
That's great Matt. If so I would need to know which LEDs you use, see your program loaded (like which example, and code tweaks done),  see how you wired it and then if needed go back and forth until it works.

mattarth, 2017-05-09:
Thanks a lot. I'm using WS28128B modules on this strip from Cool
Components:
https://www.coolcomponents.co.uk/en/digital-rgb-led-weatherproof-strip-144-led-1m-black.html

I can answer the rest of your questions later today. What's the best way of
communicating with you? Is email through GitHub fine for you?

mattarth, 2017-05-09:
I'm using the simple example ws28128b code in your library with no modifications, plugging into pin 6 on the DIGITAL PWM

sonyhome, 2017-07-12:
OMG. I completely spaced out on responding to these requests. I'm sorry Mattarth...
I don't know how to direct message you with github :)

sonyhome, 2017-07-12:
Possible fix:
=========
I believe the mega pinout (pin #) and port assignments are completely different from Uno.
Ports are named PA through PH, from 0 to 7 (PA0, PA1 etc.)

See here:
http://pighixxx.com/megapdf.pdf

To make it work for example, using that schematic:
Pick PWM pin 12 on the board.
It should be wired to pin 25 of the micro-controller,
which in turn maps logically to PORT B6.

You would then change my code example to use port B6 instead of port D6 when you declare the LEDs:
ws2812b<**B,6**> myWs2812;

So to recap:
use pin 12, and program port B6.

Note: FAB_LED knows about ports A through F. It seems I should add ports G & H too.

sonyhome, 2017-08-23:
I have confirmed that a Mega 2560 works out of the box. You just need to know the mapping between ports (like D0) and pin numbers assigned by Arduino.

I used pin D0, which translates to pin labelled "SCL 21".

McAleerElk, 2021-05-13:
> I have finally got my hands on, and **confirmed that a Mega 2560 works out of the box**. You only need to know the mapping between ports (like D0) and pin numbers assigned by Arduino.
> 
> I programmed the use of port D0, which translates to pin labelled "SCL 21", and wired the LED strip on pin 21.

Now this is awesome news for someone who is about to start with FAB_LED. 

One concern I have, that have not been addressed, which in my oppinion is 'of the essence' with this specific library.

The description says:  FAB_LED allows you to control an infinite number of LEDs even with a small 16MHz Arduino

How does this correlate with Healthyuncle 's message, stating that after 250LED's he ran out of memory? 

The reason for using this particular library seems to be: make huge LED-projects that sustains high 'FPS'.  

No one has tried, for instance, 3000 LEDS?

sonyhome, 2021-05-16:
McAleerElk, You are correct that with AVR, memory is going to be a limiting factor IF you're going to define your pixels in an array. Even with ARM or other you would have the same limitations but at a much larger practical limit that makes it moot for most.

Now how can you go to high numbers and/or infinite pixels in theory with FAB_LED?

* Use a color palette, which allows you you define arrays of pixels with fewer bytes. If you use few colors then the color palette is also small.
* Use simple color patterns that can be expressed with a fast iterative algorithm (like a rainbow) and compute the pixel color in flight as it's being bit banged into the LED strip.

That's great but an infinite number of pixels would take an infinite amount of time to bit-bang.

* FAB_LED has a mode that allows you to drive in parallel up to 6 ports, cutting the refresh latency by up to 6.
It does so by taking 6 pixels from the pixel array, extracting a bit from each and combining them into a byte that's pushed to a port. The limit of 6 is just that the code's not fast enough to push 8 (number of pins for a port).

What's the catch?

You probably can't mix n match all the above modes with FAB_LED as you'll run out of CPU cycles during the bit banging OR might hit a limit of how FAB_LED is coded and allows you to express your algorithm.

See the code examples to grok how what I described works.

jamesdelaneyie, 2022-05-30:
Hello there! Firstly thanks for the library! it's working well for my purposes so far, I previously ran into update speed issues using FastLED, so I think ditching the buffer is what I need. However! I am using a MEGA and wanted to attempt to add the other ports to your library since I need those extra pins. Am I right in saying that the areas I would need to extend would be: 

LOC: 170 — Add the extra ports in a similar manner
LOC: 222 — Add the extra ports here
LOC: 237 — Same as above mostly 
LOC: 275 — Same as above mostly
LOC: 318 — Again, copy the format and add the extra ports
LOC: 325 — Same as above 
LOC: 355 — Same as above
LOC: 362 — Same as above
LOC: 807 — Optional add to debugging script here
LOC: 834 — Same as above

sonyhome, 2022-05-30:
I believe you are correct. Basically find the port numbers and letters and add them. It's been a while so I don't remember if adding new ports can break other chips but I would expect not since I use an AtTiny85 for my small builds and it doesn't have that many ports, and it would likely have been something I would have ifdef'd.
If you do an update I can fold it in the code base.

jamesdelaneyie, 2022-06-04:
Yep worked a treat, I'm pretty bad at github but I forked the repo and added my updates here: https://github.com/jamesdelaneyie/FAB_LED/blob/master/FAB_LED.h 

not sure how to do a proper pull request so hope the above works for you!