sonyhome/FAB_LED #18: To Complicate Exampels

bobybobybob, 2017-01-02:
It would be nice for some very easy example Scripts . That are easy to understand 

For Example a single APA106 with some fixed colors
Blue, Red, Green, White, some Fading and Blinking. 

Another Example the same for manyPort 

Would be very usefull for all who dont use strips and want to use APA106 LED only for Status Lamp and saving some Digital Ports (one old RGB LED need 3)  with the APA106 only one. 

Thank you and Best Regards

sonyhome, 2017-01-22:
OK I will look into making very simple examples.

The examples I have made are meant to validate the code, more so than to show how to use.

sonyhome, 2017-01-27:
I added the 0_simpleExamples hopefully simple enough!

closed 2017-01-27 by sonyhome

bobybobybob, 2017-01-28:
Thank you very much