sonyhome/FAB_LED #20: Compiling errors

danki, 2017-07-16:
Ive tried various adaptions right in sources, replacing "\" with spaces (suspect those are used for compiler to see 2 line code on 1 line) but after each modification i get new errors.

For starters, im using Uno and Pro mini but wont get compiler to work for neither of them,
heres a output...

**Any ideas why ??**

`
In file included from /Users/deko/Documents/Arduino/libraries/FAB_LED-master/Examples/0_simpleExamples/sk6812b/sk6812b.ino:31:0:
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h: In static member function 'static void avrBitbangLedStrip<high1, low1, high0, low0, minMsRefresh, dataPortId, dataPortPin, clockPortId, clockPortPin, colors, protocol>::sendPixels(uint16_t, const rgbw*)':
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1266:126: error: 'else' without a previous 'if'
 #define SEND_REMAPPED_PIXELS(color, numPixels, array) if IS_PIXEL_FORMAT_3B(color) SEND_REMAPPED_PIXELS_3B(numPixels, array) else SEND_REMAPPED_PIXELS_4B(numPixels, array)
                                                                                                                              ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1284:3: note: in expansion of macro 'SEND_REMAPPED_PIXELS'
   SEND_REMAPPED_PIXELS(colors, numPixels, array);
   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h: In static member function 'static void avrBitbangLedStrip<high1, low1, high0, low0, minMsRefresh, dataPortId, dataPortPin, clockPortId, clockPortPin, colors, protocol>::sendPixels(uint16_t, const grbw*)':
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1266:126: error: 'else' without a previous 'if'
 #define SEND_REMAPPED_PIXELS(color, numPixels, array) if IS_PIXEL_FORMAT_3B(color) SEND_REMAPPED_PIXELS_3B(numPixels, array) else SEND_REMAPPED_PIXELS_4B(numPixels, array)
                                                                                                                              ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1303:3: note: in expansion of macro 'SEND_REMAPPED_PIXELS'
   SEND_REMAPPED_PIXELS(colors, numPixels, array);
   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h: In static member function 'static void avrBitbangLedStrip<high1, low1, high0, low0, minMsRefresh, dataPortId, dataPortPin, clockPortId, clockPortPin, colors, protocol>::sendPixels(uint16_t, const hbgr*)':
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1266:126: error: 'else' without a previous 'if'
 #define SEND_REMAPPED_PIXELS(color, numPixels, array) if IS_PIXEL_FORMAT_3B(color) SEND_REMAPPED_PIXELS_3B(numPixels, array) else SEND_REMAPPED_PIXELS_4B(numPixels, array)
                                                                                                                              ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1322:3: note: in expansion of macro 'SEND_REMAPPED_PIXELS'
   SEND_REMAPPED_PIXELS(colors, numPixels, array);
   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h: In static member function 'static void avrBitbangLedStrip<high1, low1, high0, low0, minMsRefresh, dataPortId, dataPortPin, clockPortId, clockPortPin, colors, protocol>::sendPixels(uint16_t, const rgb*)':
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1266:126: error: 'else' without a previous 'if'
 #define SEND_REMAPPED_PIXELS(color, numPixels, array) if IS_PIXEL_FORMAT_3B(color) SEND_REMAPPED_PIXELS_3B(numPixels, array) else SEND_REMAPPED_PIXELS_4B(numPixels, array)
                                                                                                                              ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1338:3: note: in expansion of macro 'SEND_REMAPPED_PIXELS'
   SEND_REMAPPED_PIXELS(colors, numPixels, array);
   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1263:47: error: 'const rgb' has no member named 'w'
 #define sendWhiteMacro sendBytes(1, &array[i].w)
                                               ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1243:6: note: in definition of macro '_SEND_REMAPPED_PIXELS'
      sendWhite;                 \
      ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1264:91: note: in expansion of macro 'sendWhiteMacro'
 #define SEND_REMAPPED_PIXELS_4B(numPixels, array) _SEND_REMAPPED_PIXELS(numPixels, array, sendWhiteMacro)
                                                                                           ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1266:131: note: in expansion of macro 'SEND_REMAPPED_PIXELS_4B'
 #define SEND_REMAPPED_PIXELS(color, numPixels, array) if IS_PIXEL_FORMAT_3B(color) SEND_REMAPPED_PIXELS_3B(numPixels, array) else SEND_REMAPPED_PIXELS_4B(numPixels, array)
                                                                                                                                   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1338:3: note: in expansion of macro 'SEND_REMAPPED_PIXELS'
   SEND_REMAPPED_PIXELS(colors, numPixels, array);
   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1263:47: error: 'const rgb' has no member named 'w'
 #define sendWhiteMacro sendBytes(1, &array[i].w)
                                               ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1249:6: note: in definition of macro '_SEND_REMAPPED_PIXELS'
      sendWhite;                 \
      ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1264:91: note: in expansion of macro 'sendWhiteMacro'
 #define SEND_REMAPPED_PIXELS_4B(numPixels, array) _SEND_REMAPPED_PIXELS(numPixels, array, sendWhiteMacro)
                                                                                           ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1266:131: note: in expansion of macro 'SEND_REMAPPED_PIXELS_4B'
 #define SEND_REMAPPED_PIXELS(color, numPixels, array) if IS_PIXEL_FORMAT_3B(color) SEND_REMAPPED_PIXELS_3B(numPixels, array) else SEND_REMAPPED_PIXELS_4B(numPixels, array)
                                                                                                                                   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1338:3: note: in expansion of macro 'SEND_REMAPPED_PIXELS'
   SEND_REMAPPED_PIXELS(colors, numPixels, array);
   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1263:47: error: 'const rgb' has no member named 'w'
 #define sendWhiteMacro sendBytes(1, &array[i].w)
                                               ^
[... 95 lines cut ...]
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1266:131: note: in expansion of macro 'SEND_REMAPPED_PIXELS_4B'
 #define SEND_REMAPPED_PIXELS(color, numPixels, array) if IS_PIXEL_FORMAT_3B(color) SEND_REMAPPED_PIXELS_3B(numPixels, array) else SEND_REMAPPED_PIXELS_4B(numPixels, array)
                                                                                                                                   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1364:3: note: in expansion of macro 'SEND_REMAPPED_PIXELS'
   SEND_REMAPPED_PIXELS(colors, numPixels, array);
   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1263:47: error: 'const bgr' has no member named 'w'
 #define sendWhiteMacro sendBytes(1, &array[i].w)
                                               ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1252:6: note: in definition of macro '_SEND_REMAPPED_PIXELS'
      sendWhite;                 \
      ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1264:91: note: in expansion of macro 'sendWhiteMacro'
 #define SEND_REMAPPED_PIXELS_4B(numPixels, array) _SEND_REMAPPED_PIXELS(numPixels, array, sendWhiteMacro)
                                                                                           ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1266:131: note: in expansion of macro 'SEND_REMAPPED_PIXELS_4B'
 #define SEND_REMAPPED_PIXELS(color, numPixels, array) if IS_PIXEL_FORMAT_3B(color) SEND_REMAPPED_PIXELS_3B(numPixels, array) else SEND_REMAPPED_PIXELS_4B(numPixels, array)
                                                                                                                                   ^
/Users/deko/Documents/Arduino/libraries/FAB_LED-master/FAB_LED.h:1364:3: note: in expansion of macro 'SEND_REMAPPED_PIXELS'
   SEND_REMAPPED_PIXELS(colors, numPixels, array);
   ^
exit status 1
Error compiling for board Arduino/Genuino Uno.

`

sonyhome, 2017-07-17:
Hello,

Thanks for notifying me of this problem.

Can you post your source code too, or if it is one of my examples, which one this occurs with? I will look into this!

sonyhome, 2017-07-17:
Looks like my last commit was hasty. Please use the previous code change for your testing while I find time to fix it, it should work.

I believe this would be the link:
https://github.com/sonyhome/FAB_LED/tree/9a4d5dfd7c742af2f0f30ce9531530aa4271823c

sonyhome, 2017-07-18:
Fixed with July 18 commit.

Note: Compile error is solved. However I can't test the code on hardware with LEDs right now to verify everything else is OK.

If there are more problems, please reopen/comment on this bug.

closed 2017-07-18 by sonyhome