sonyhome/FAB_LED #31: Doesn't work on ATmega328 @ 8mhz with internal oscillator

aaronaverill, 2019-08-18:
I'm using a bare ATmega328 board with the internal oscillator fuse set running at 8mhz. The FAB_LED library doesn't work. I've tried 3 other libraries all coded using assembly (light_ws2812, Adafruit_NeoPixel, FastLED) and all work properly on the same chip with the same settings. I've compiled in both Arduino and in GNU-AVR. It seems the code generation isn't correct. A pity, I really enjoy using this on the ATtiny85 and it works great at 16mhz. 

Probably the documentation needs to be updated.

sonyhome, 2019-09-06:
Hello,

Running at 8MHz would require me to do some experimenting, as the timings at that frequency is rather tight. It has to do with the choice of doing bit-banging with no assembly code. I now have an oscilloscope to tune the code but I lack the time.

The goal of this library is to be nimble and memory efficient and hence I should fix that. :) There is no ETA right now, and I may do a rewrite at some point with a better code structure.

aaronaverill, 2019-09-10:
I didn't intent that you would add 8mhz support, just to update the documentation so that people are aware of the library limitations, this section in the README.md file.

That makes it portable to Arduino boards of any frequency 8MHz and above (8MHz is untested). 

Should be corrected to 16Mhz, or minimally, that it doesn't operate on the atmega328, which is a very common microcontroller.

closed 2021-09-08 by sonyhome