sonyhome/FAB_LED #32: Library is Not comfortable for newbies

S60Team, 2020-04-14:
Please, make library **more comfortable**: write functions, like in NeoPixel library: they are simple to use.
Like: _strip.clear(), strip.fill(255,255,255), strip.show(), strip.setPixelColor(pixel, 255,255,255)_

sonyhome, 2020-04-17:
Thank you for your feedback:
I'll think about it, but this library separates the array holding the pixels from the port the pixels are written to. Maybe I can extend the pixel classes for that.
For Adafruit's library both are managed at the same time which is much less flexible so clear/fill/set make sense, but here it doesn't as much. You can clear the LEDs with FAB_LED but that doesn't erase the array.

S60Team, 2020-04-17:
Thank you for your response. Then I think, your lib needs more code examples. And classes, whatever it means :)

sonyhome, 2020-04-19:
Right now I'm working on a DigitalIO library (for sensors connected on the digital pins).

That one will be easier to use like you say, and support a few things I feel would make the sensors I know and have plug and play (like knock, rotary and ultrasonic sensors).

closed 2021-09-08 by sonyhome