sonyhome/FAB_LED #37: Need program for stair led project.

a619sachin, 2021-09-08:
I need example for using two PIR motion sesors and WS2812B led (600 pixels in total). I tried Adafruit library and I was unable to go beyond 548 leds using Arduino Uno. One member in Arduino forum mentioned about this library but I cannot find sufficient example for my setup to work.

ednieuw, 2021-09-08:
Use the €15 Arduino nano Every. It has more memory.

a619sachin, 2021-09-08:
@ednieuw I think they both have EEPROM/SRAM [kb] 1/2 and Flash Memory 32kb, right!!

ednieuw, 2021-09-08:
The nano every has 48kb memory. I use it with 625 Sk6812 leds.
It is also cheap and can be bought as a original Arduino product at their website

a619sachin, 2021-09-08:
@ednieuw I get this message after compiling the code
Sketch uses 4578 bytes (14%) of program storage space. Maximum is 32256 bytes.
Global variables use 230 bytes (11%) of dynamic memory, leaving 1818 bytes for local variables. Maximum is 2048 bytes.
Do you think there is memory problem?

ednieuw, 2021-09-08:
You can check it by reducing the amount of leds to lets day 10 leds. If the compiler reports the same amount of free memory you know it is not reporting the memory that is allocated for the leds. 
With the adafruit library this was the case with my program. In my case I used 100 leds and the program size was 23000 bytes. With more leds the program went havoc. 
You can also try another ws2812 library like

https://github.com/cpldcpu/light_ws2812

a619sachin, 2021-09-08:
I commented all the debug helping line. It worked

a619sachin, 2021-09-08:
@sonyhome please delete this issue. I think it's not relevant to current library.

closed 2021-09-08 by a619sachin

sonyhome, 2021-09-08:
Thank you for closing this request.

For info:
Each LED uses about 3 bytes of memory for storage of the display array, like Adafruit's unless you go into the other modes (color palette or on the fly computation of the LED colors, depending on the pattern complexity). This library is more memory savvy than Adafruit's but it will only get you so far.

When a program runs it also needs a stack to call and return from functions, which may overrun your free memory and is not accounted for. The only way to know if it fits in a small memory footprint is to try it out.